agilexrobotics/piper_sdk #2: windows下的控制问题

Dogplanningloyo, 2024-11-04:
在windows下，即使使用python-can，也无法找到对应的端口进行控制，可否提供windows下的机械臂控制demo

WangYuLin-SEU, 2024-11-27:
> 在windows下，即使使用python-can，也无法找到对应的端口进行控制，可否提供windows下的机械臂控制demo

windows下需要安装can的依赖项后才能安装python-can，可以看一个CSDN的教程：https://blog.csdn.net/X571938/article/details/135989064。
此外，我觉得，官方应该做一个通用于linux、max、windows的demo，例如用CANtact作为python-can的依赖项，而不是做一个非常以来linux环境的demo。

fengsvkn, 2025-05-19:
> > 在windows下，即使使用python-can，也无法找到对应的端口进行控制，可否提供windows下的机械臂控制demo
> 
> windows下需要安装can的依赖项后才能安装python-can，可以看一个CSDN的教程：[https://blog.csdn.net/X571938/article/details/135989064。](https://blog.csdn.net/X571938/article/details/135989064%E3%80%82) 此外，我觉得，官方应该做一个通用于linux、max、windows的demo，例如用CANtact作为python-can的依赖项，而不是做一个非常依赖linux环境的demo。
> 
> 2024-12-30：我已经完成了基于cantact的win版本的piper-sdk，目前是能够正确读取机械臂的关节信息，其他功能还有待于进一步测试。

可以提供一个运行历程吗？

WangYuLin-SEU, 2025-06-16:
> 可以提供一个运行历程吗？

最近很忙碌，没抽出时间回复，以下是在win10上使用的大致流程。

首先，你需要安装cantact；

然后，你需要修改piper_interface.py以及can_encapsulation.py中关于can初始化的代码，如下：

[image]

[image]

[image]

祝你顺利！

supperyun, 2026-01-15:
> 最近很忙碌，没抽出时间回复，以下是在win10上使用的大致流程。
>
> 首先，你需要安装cantact；
>
> 然后，你需要修改piper_interface.py以及can_encapsulation.py中关于can初始化的代码，如下：
>
> [image]
>
> [image]
>
> [image]
>
> 祝你顺利！

感谢大佬，还没采购piper，就是担心Windows控制不行，我其他设备的控制全部在windows上，转移比较麻烦。请问您的sdk改造可以实现信息读取可机械臂控制了吗

WangYuLin-SEU, 2026-01-15:
> > > > > > > 在windows下，即使使用python-can，也无法找到对应的端口进行控制，可否提供windows下的机械臂控制demo
> > > > > > 
> > > > > > 
> > > > > > windows下需要安装can的依赖项后才能安装python-can，可以看一个CSDN的教程：[https://blog.csdn.net/X571938/article/details/135989064。](https://blog.csdn.net/X571938/article/details/135989064%E3%80%82) 此外，我觉得，官方应该做一个通用于linux、max、windows的demo，例如用CANtact作为python-can的依赖项，而不是做一个非常依赖linux环境的demo。
> > > > > > 2024-12-30：我已经完成了基于cantact的win版本的piper-sdk，目前是能够正确读取机械臂的关节信息，其他功能还有待于进一步测试。
> > > > > 
> > > > > 
> > > > > 可以提供一个运行历程吗？
> > > > 
> > > > 
> > > > 最近很忙碌，没抽出时间回复，以下是在win10上使用的大致流程。
> > > > 首先，你需要安装cantact；
> > > > 然后，你需要修改piper_interface.py以及can_encapsulation.py中关于can初始化的代码，如下：
> > > > [image]
> > > > [image]
> > > > [image]
> > > > 祝你顺利！
> > > 
> > > 
> > > 感谢大佬，还没采购piper，就是担心Windows控制不行，我其他设备的控制全部在windows上，转移比较麻烦。请问您的sdk改造可以实现信息读取可机械臂控制了吗
> > 
> > 
> > 我在旧版本上是能用的，新版本的就不太行了。目前，还是用的nvidia的开发板，自带的linux系统来调用piper。
> 
> 您使用的jetson orin开发板吗？我也在设想能否使用jetson设备搭建api服务作为桥梁，然后windows调用api实现控制piper。请问jetson调用piper能否稳定跑起来呀

非常稳定，我用的是nano，两三千的那种。