agilexrobotics/piper_sdk #3: End Effector coordinates?

haixuanTao, 2024-11-13:
hello,

Would it be possible to have end effector coordinates for both receiving and sending positions?

Thanks in advance.

RosenYin, 2024-11-19:
你可以使用GetArmEndPoseMsgs获取末端位姿，EndPoseCtrl给出目标位姿

haixuanTao, 2024-11-19:
Ah ok!

没有看到： 
```python
    def EndPoseCtrl(self,X,Y,Z,RX,RY,RZ):
        self.__CartesianCtrl_XY(X,Y)
        self.__CartesianCtrl_ZRX(Z,RX)
        self.__CartesianCtrl_RYRZ(RY,RZ)
```

Thanks

haixuanTao, 2024-11-19:
```python
time stamp:0
ArmMsgEndPoseFeedBack:
 X_axis : 0, 0.000
 Y_axis : 0, 0.000
 Z_axis : 0, 0.000
 RX_axis : 0, 0.000
 RY_axis : 0, 0.000
 RZ_axis : 0, 0.000

```
Actually it seems that I get no message on `GetArmEndPoseMsgs()`

haixuanTao, 2024-11-20:
So it seems that sometimes, the can bus does not work and i can't receive message anymore.

RosenYin, 2024-11-20:
机械臂状态、机械臂关节角、末端位姿、机械臂关节限位等消息是默认都接收的（如果没有说明是应答式消息的话），如果没有接收到消息，检查线路以及机械臂指示灯，重新上电再尝试

nguyenanhtuan1008, 2024-12-09:
@RosenYin Could you tell me how to work with EndPoseCtrl(X, Y, Z, Rx, Ry, Rz)? I try to input example EndPoseCtrl(10000, 20000, 30000, 40000, 50000, 60000) but it didn't work. i want to do the move C

Ke-Wang1017, 2024-12-13:
> @RosenYin Could you tell me how to work with EndPoseCtrl(X, Y, Z, Rx, Ry, Rz)? I try to input example EndPoseCtrl(10000, 20000, 30000, 40000, 50000, 60000) but it didn't work. i want to do the move C

Hi you need to change the control mode, as shown here https://github.com/agilexrobotics/piper_sdk/blob/master/demo/piper_moveC.py

RosenYin, 2024-12-18:
我已经在新的提交中的demo中增加了movec的demo，你可以尝试运行，但注意，由于欧拉角死锁问题，给定的最终目标姿态，和反馈的不太一样，但是姿态是一样的

nguyenanhtuan1008, 2024-12-18:
@RosenYin 
Thank you very much. I will have a try in the next couple of days!

closed 2025-03-31 by HoranTian