agilexrobotics/piper_sdk #8: End pose angle feedback jumps

Ke-Wang1017, 2024-12-13:
HI I am running the https://github.com/agilexrobotics/piper_sdk/blob/master/demo/piper_joint_ctrl.py and get the endpose through `piper.GetArmEndPoseMsgs()` and recorded the data. After plotting I find that the angle is jumping, I think this might because the conversion from quaternion to euler angles is not done properly. 
Plots has been shown below:
[image]

RosenYin, 2024-12-17:
piper机械臂底层固件暂时没有四元数接口，只反馈了欧拉角，sdk只是将底层的数据转发出来，这个问题和底层固件有关，请等待固件更新

Ke-Wang1017, 2025-01-16:
> piper机械臂底层固件暂时没有四元数接口，只反馈了欧拉角，sdk只是将底层的数据转发出来，这个问题和底层固件有关，请等待固件更新

你们有没有开放接口我们可以自己做forward kinametics 啊？我现在知道了各个关节的角度，我自己计算末端的位置和方向
,这个数据记录没办法用来训练啊

RosenYin, 2025-02-19:
最新的0.1.14版本sdk已经增加正解，可以看下release的log

closed 2025-02-20 by Ke-Wang1017