agilexrobotics/piper_sdk #10: How to increase speed of JointCtrl for remote teleoperation.

boyentenbi, 2025-01-02:
Hi AgileX devs, I am teleoperating my Piper arms *over a remote connection*, and I'd like to increase the responsiveness.

I have a Gello-like controller that I use to generate inputs (I'm not using a Piper arm to do this). I use JointCtrl at high frequency to update the desired joint position. I am *not* using EndPoseCtrl.

At the moment, the speed is much worse than what I see in your videos. I think this is because the normal (local) method connects the master and slave arms directly through the firmware. 

I have tried to increase the speed as follows:
* MotorAngleLimitMaxSpdSet (this clearly works because when I set it to a low value the arm stops moving, but increasing it to a very high value doesn't help)
* EndSpdAndAccParamSet (seems to have no effect, maybe this is only used for EndPoseCtrl?)
* MIT mode: MotionCtrl_2(0, 0, 0, 0xAD). This makes the motion extremely fast, but very jerky. 

Honestly, my ideal setup would simply be to see the firmware myself, implement a simple PID controller and tune the gains. is it possible to get source access to the firmware and some notes on how to build it?

Failing that, would you be able to provide some advice on how to get the same speed as the built in master-slave teleoperation using JointCtrl? 

Thank you!

RosenYin, 2025-01-03:
当设置为主从模式的时候，MotionCtrl_2（0，0，0，0xAD）就是设置从臂的指令，这样会令响应加快，如果要自己设定pid，请等待后续开放电机驱动器底层控制

boyentenbi, 2025-01-03:
Ah okay, so the normal master/slave mode uses MotionCtrl_2(..., 0xAD), thank you! I think I made a mistake with my previous attempt at this because using MotionCtrl_2(...,0xAD) plus JointCtrl with my GELLO controller is very responsive!

Note that I have to use MotionCtrl_2(0x01, 0x01, 0, 0xAD), not self.piper.MotionCtrl_2(0, 0, 0, 0xAD). The latter causes the arm to not move.

closed 2025-01-03 by boyentenbi

prakharg01, 2025-02-04:
> When set to master-slave mode, MotionCtrl_2 (0, 0, 0, 0xAD) is the command to set the slave arm, which will speed up the response. If you want to set the PID yourself, please wait for the subsequent release of the motor driver bottom-level control

Is there any update on the release of the low level motor control directly?

RosenYin, 2025-02-07:
已经更新0.1.12，请查看更新日志，路径在asserts/V2/CHANGELOG-V2.MD中，这个文件并没有显式的在README中标明，因为其中的MIT控制驱动器部分被认为是危险的，需要谨慎使用