agilexrobotics/piper_sdk #11: Teach mode through programming interface

trzy, 2025-01-05:
Hi,

I was recently informed in a support chat on WhatsApp that it should be possible to place the robot into "teach mode" (the same mode as when the button on the robot is pressed). That is, I would like to be able to programmatically enter this mode that allows the user to freely position the robot while it holds its position, and read back the various joint angles. Sometimes this is referred to as "gravity compensated" operation.

I was told that MIT mode should achieve this:

```
piper.MotionCtrl_2(ctrl_mode=0x01, move_mode=0x01, move_spd_rate_ctrl=50, is_mit_mode=0xad)
```

However, this does not have any effect (other than the joint 5 moving slightly when the mode is first entered).

Is it possible to enter this mode? I would like joints 1-6 to be in gravity compensated mode, teachable by a human, and the gripper to continue to be operated by the program (it is controlled separately anyway so that should be simple).

Thank you,

Bart

RosenYin, 2025-01-06:
在当前固件下，按键示教模式只能通过按键触发，因为底层固件的逻辑是按键优先级最高，现阶段是没有通过协议来修改机械臂模式到按键示教模式的

trzy, 2025-01-06:
I see -- will the firmware be updated to allow this mode to be triggered in software? Alternatively, 1) can the joint values be read while the robot is in this mode (after the user physically presses the button), and 2) can the gripper motor still be controlled via software while in this mode? A workaround for me would be to press the button, collect joint data as the user performs a task with the arm, and also be able to operate the gripper in software (rather than having the user open and close the gripper with their hands, I want to mount a slider control to the robot that transmits back to the computer and controls the gripper in a more intuitive way).

Thank you.

trzy, 2025-01-11:
The joints can be read in teach mode, I have confirmed. However, it would still be incredibly useful to allow the gripper to be controlled through software while joints 1-6 are in teach mode (even if activated by the external button). Would it be difficult to add this capability in a firmware update? Operating the gripper through software when in teach mode (to collect demonstrations) would be very useful. It's easy to manipulate the arm with two hands but difficult to manipulate the gripper (and impossible to capture the force necessary to pick up certain objects). Being able to use software (e.g., I can hook up the gripper control to an iPhone app or add my own physical button that communicates with the host PC to control the gripper) would make this all much easier.

Thank you.

Ke-Wang1017, 2025-01-14:
那能不能按键模式下能用软件或者什么方式控制夹具？ 如果这样的话我们通过手拖也能示教抓取东西，其实是个很好用的feature

Ke-Wang1017, 2025-01-14:
> 在当前固件下，按键示教模式只能通过按键触发，因为底层固件的逻辑是按键优先级最高，现阶段是没有通过协议来修改机械臂模式到按键示教模式的

那能不能按键模式下能用软件或者什么方式控制夹具？ 如果这样的话我们通过手拖也能示教抓取东西，其实是个很好用的feature

sxy-sun, 2025-02-03:
> 在当前固件下，按键示教模式只能通过按键触发，因为底层固件的逻辑是按键优先级最高，现阶段是没有通过协议来修改机械臂模式到按键示教模式的

Hi, I am wondering if the teaching mode can only be triggered by the button, what does this do when ctrl_mode set to 0x02

```python
def MotionCtrl_2(self, ctrl_mode, move_mode, move_spd_rate_ctrl, is_mit_mode=0x00))
   '''
        ctrl_mode: 控制模式 uint8 
            - 0x02 示教模式
   '''
```
Also here when grag_teach_ctrl set to 0x01
```python
def MotionCtrl_1(self, emergency_stop, track_ctrl, grag_teach_ctrl)
    '''
        grag_teach_ctrl: 拖动示教指令 uint8 0x00 关闭
            - 0x01 开始示教记录（进入拖动示教模式）
            - 0x02 结束示教记录（退出拖动示教模式） 
            - 0x03 执行示教轨迹（拖动示教轨迹复现） 
            - 0x04 暂停执行 
            - 0x05 继续执行（轨迹复现继续） 
            - 0x06 终止执行 
            - 0x07 运动到轨迹起点
    '''
```

RosenYin, 2025-02-19:
这个0x02经过测试是无效值，在最新版本sdk中已经去除

wtingsamDenso, 2025-07-11:
I think this could be related the topic, so I left a comment here. 
Before I started teaching mode, I set my program to wait. After completing the teaching process using the button, I pressed the button again to confirm that teaching mode was properly exited (the green light turned off as expected).
However, when I resumed the program afterward, I found that the PiPER arm was unresponsive or out of control.
Could you please help me troubleshoot this issue?

Thank you in advance!

maxpolzin, 2025-07-22:
Same here. After finishing the teaching mode, the Piper (python3 detect_arm.py) continues indicating that it is still in teaching mode.