agilexrobotics/piper_sdk #20: Arm freezing occasionally (under high speed / acceleration?)

boyentenbi, 2025-02-17:
Hi, I've been using the piper arms for some time now. We observe that the arm freezes itself under certain conditions. I think this happens when the arm has a high acceleration or speed. For what it's worth, I am using the arm in 'MIT mode'.

**Can you help me understand why this is the case and if there is a configuration parameter to change when this happens?**

Things I have tried:

I can see in piper_interface.py that there is max angle / speed / acc limits. However, I think these are limits on the commanded values rather than the values experienced by the arm. Using `SearchAllMotorMaxAccLimit() -> piper.piper.GetAllMotorMaxAccLimit()` and `SearchAllMotorMaxAngleSpd() -> GetAllMotorAngleLimitMaxSpd()` I get the following output:

Motor Limits:
| Motor | Angle Limit          | Speed Limit   | Accel Limit   |
|-------|------------------------|---------------|---------------|
| 0     | 0, 0                     | 0             | 0             |
| 1     | -1500, 1500              | 300           | 500           |
| 2     | 0, 1800                  | 300           | 500           |
| 3     | -1700, 0                 | 320           | 500           |
| 4     | -1000, 1000              | 300           | 500           |
| 5     | -700, 700                | 300           | 500           |
| 6     | -1800, 1800              | 300           | 500           |

Also, when I print GetCrashProtectionLevelFeedback, it says that it has value 0 (no protection) for all joints.

boyentenbi, 2025-02-18:
Some more observations: I am confident that this happens when J2 is under heavy load (e.g when the arm is fully extended and rapidly accelerating / decelerating). Any assistance in how to prevent this from happening, or how to re-enable the arm after it has frozen would be helpful.

RosenYin, 2025-02-19:
有一种更安全的令机械臂停止的指令，是`MotionCtrl_1(0x01, 0, 0)`，这个指令是机械臂急停指令，执行后机械臂会以一个阻尼缓慢下落而不是立刻断电下落，在机械臂缓慢掉落到最低点后(注意，进入到急停模式后，任何指令都无效，想要继续控制机械臂必须执行reset指令)，您需要使用`MotionCtrl_1(0x02, 0, 0)`来reset机械臂，然后再设定机械臂模式为mit或者是位置速度控制模式，后续我们会将这个demo增加进去，也会添加到我们piper的qt的ui上，到时候您可以使用ui来stop机械臂。

agilexrobotics, 2025-02-19:
To translate Rosen's answer: 
There is a safer command to stop the robotic arm, which is MotionCtrl_1(0x01, 0, 0). This command is an emergency stop command for the robotic arm. After execution, the robotic arm will slowly fall down with a damping effect instead of immediately dropping off like turning off the power. 

After the robotic arm slowly drops to the lowest point (note that after entering the emergency stop mode, any command is invalid. If you want to continue to control the robotic arm, you must execute the reset command), you need to use MotionCtrl_1(0x02, 0, 0) to reset the robotic arm and then set the robotic arm mode to MIT or position and speed control mode. 

After the robotic arm slowly drops to the lowest point (note that after entering the emergency stop mode, any command is invalid. If you want to continue to control the robotic arm, you must execute the reset command), you need to use MotionCtrl_1(0x02, 0, 0) to reset the robotic arm and then set the robotic arm mode to MIT or position and speed control mode.

closed 2025-03-11 by RosenYin