agilexrobotics/piper_sdk #24: Agilex Piper CAN does not work with MacOS

cetiennec, 2025-02-25:
Hello,

Thank you for this great product!

I have been trying to get the piper sdk running with MacOS?

### Diagnostic: USB plugged, no device detected

The following allowed me to detect the candlelight CAN : 

`system_profiler SPUSBDataType`

With associated answer : 

```
USB:

    USB 3.1 Bus:

      Host Controller Driver: AppleT8122USBXHCI

        candleLight USB to CAN adapter:

          Product ID: 0x606f
          Vendor ID: 0x1d50
          Version: 0.00
          Serial Number: 003900305246571120393733
          Speed: Up to 12 Mb/s
          Manufacturer: bytewerk
          Location ID: 0x02100000 / 1
          Current Available (mA): 500
          Current Required (mA): 150
          Extra Operating Current (mA): 0
```

But when running

` ls /dev | grep -E 'cu\.|tty\.'`

To see which port I can use, there is no trace of the CAN:

`cu.BIMBAMBOOM
cu.Bluetooth-Incoming-Port
cu.wlan-debug
tty.BIMBAMBOOM
tty.Bluetooth-Incoming-Port
tty.wlan-debug`

(those are my personal devices)

## Potential solutions

For now, I'm investigating the following leads : 

### 1. Docker

Runningt a docker instance, based on ubuntu 22:04 and the following instructions : 
```
RUN apt-get update && \
    apt-get install -y --no-install-recommends \
    python3 python3-pip can-utils ethtool iproute2 iputils-ping \
    && rm -rf /var/lib/apt/lists/*

# Install python-can and piper_sdk
RUN pip3 install --no-cache-dir python-can piper_sdk
```

I kept having the following error despite multiple docker configuration : 

```
> Error: The number of CAN modules detected by the system (0) does not match the expected number (1).
```

### 2. MacOS CAN tools

I tried installing can-utils and eth tools equivalent on mac and modified the can_activate.sh accordingly. I tested :

- https://github.com/carloop/can-utils-osx
- https://www.mac-can.com
- https://canable.io/getting-started.html 

But the device was not detected. (same kind of error: no device connected). I

### 3. Other hardware

I considered changing the CAN-board for one with mac OS-provided driver but not sure of which one could work. What alternative hardware do you recommend for MacOS compatibility ? 

Who do you think I should reach out to for MacOS CAN compatibility ? 

I'd like to make more tests, but I'm afraid of breaking the Agilex Piper CAN adapter and my robot.

oulianov, 2025-02-28:
I'm facing the same issue.

RosenYin, 2025-03-03:
我手里没有MacOS设备，如果可以的话，可以尝试下远程你的电脑看一下

oulianov, 2025-03-03:
Sure, we can do that together. 
You can book some time with me here : https://cal.com/nicolas-oulianov
I can share my screen and run the commands you tell me.

Crypto69, 2025-03-23:
I had to use parallels and install ubuntu v22.04

oulianov, 2025-03-26:
so it works in a VM! that's cool. Maybe there is chance it will work in a docker container 🤔  Did you try that ?

Crypto69, 2025-03-27:
I did not try this. What would you run in the container?

I wish we could get it running natively on Mac. 😞

RosenYin, 2025-07-09:
> Similar issue on ubuntu 24.04. Since ros2 is not well supported on 24.04, I run a ros-humble docker container on the laptop.ubuntu 24.04 上也有类似的问题。由于 ros2 在 24.04 上没有得到很好的支持，因此我在笔记本电脑上运行了一个 ros-humble docker 容器。
> 
> I add `--network host --device /dev/bus/usb/:/dev/bus/usb/` when run the container, I can find `candleLight USB to CAN adapter`.我在运行容器时添加 `--network host --device /dev/bus/usb/:/dev/bus/usb/` ，我可以找到 `candleLight USB to CAN adapter` .

这样能正常控制吗

RosenYin, 2025-10-30:
MacOs现在已经支持，不同的是，需要使用串口can模块而不是官方的socketcan模块，这一部分内容可以在Q&A中查看

---
macOS now supports this, but the difference is that you need to use the serial CAN module instead of the official SocketCAN module. This information can be found in the Q&A section.