agilexrobotics/piper_sdk #25: editable mode install

ronjian, 2025-02-26:
Please add the setup.py, so that I can install by editable mode (pip install -e .).

Thank you for your nice work.

RosenYin, 2025-02-26:
使用release里的tar.gz文件，里面有setup.py

ronjian, 2025-02-26:
感谢。
另外，我在源代码安装的时候，报错
```  ValueError: invalid pyproject.toml config: `project`.
  configuration error: `project` must not contain {'long_description', 'long_description_content_type'} properties```
是不是我的setuptools或pip版本新了，我的setuptools版本是75.3.0，pip版本是20.0.2.
@RosenYin 你那边能否测试一下```pip install -e .```?

RosenYin, 2025-02-28:
我更改了toml文件，现在可以用setup.py来安装

ronjian, 2025-03-04:
感谢回复。不过我拉取最新代码后，用```pip install -e .```安装项目还是有一个报错，麻烦看一下：
```
rrrr@rrrr-ThinkPad-E14:piper_sdk$ pip install -e .
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.tuna.tsinghua.edu.cn/simple
Obtaining file:///home/rrrr/workspace/cloth_folding/piper_sdk
  Installing build dependencies ... done
  Checking if build backend supports build_editable ... done
  Getting requirements to build editable ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build editable did not run successfully.
  │ exit code: 1
  ╰─> [14 lines of output]
      error: Multiple top-level packages discovered in a flat-layout: ['base', 'demo', 'asserts', 'monitor', 'protocol', 'interface', 'piper_msgs', 'kinematics', 'hardware_port'].
      
      To avoid accidental inclusion of unwanted files or directories,
      setuptools will not proceed with this build.
      
      If you are trying to create a single distribution with multiple packages
      on purpose, you should not rely on automatic discovery.
      Instead, consider the following options:
      
      1. set up custom discovery (`find` directive with `include` or `exclude`)
      2. use a `src-layout`
      3. explicitly set `py_modules` or `packages` with a list of names
      
      To find more information, look for "package discovery" on setuptools docs.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build editable did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

RosenYin, 2025-03-05:
你需要在setup.py所在的目录下执行`pip3 install .`，而不是在`pyproject.toml`文件所在的路径下

ronjian, 2025-03-05:
注释掉这一行后，完成可编辑的安装
https://github.com/agilexrobotics/piper_sdk/blob/34b06204463d616c1e13dea7cc911bda60a0c563/__init__.py#L16
谢谢

closed 2025-03-05 by ronjian

RosenYin, 2025-03-05:
> 注释掉这一行后，完成可编辑的安装
> 
> [piper_sdk/__init__.py](https://github.com/agilexrobotics/piper_sdk/blob/34b06204463d616c1e13dea7cc911bda60a0c563/__init__.py#L16)
> 
> Line 16 in [34b0620](/agilexrobotics/piper_sdk/commit/34b06204463d616c1e13dea7cc911bda60a0c563)
>  'C_PiperParserBase', 
> 
> 谢谢

这个是因为导入的包缺了，我在最新的提交里修改了init.py