agilexrobotics/piper_sdk #27: Two arms on a Jetson Orin Nano

notmahi, 2025-03-01:
Hi, I am trying to attach two arms on a Jetson Orin Nano, but over the USB-to-CAN interface, it can only detect one at a time. Is there a way to run two piper arms from a single Jetson Orin Nano device? I have successfully done so from a spare NUC we had lying around, so the arms are fine.

Update: I also found this documentation: [Jetson Docs](https://docs.nvidia.com/jetson/archives/r35.4.1/DeveloperGuide/text/HR/ControllerAreaNetworkCan.html) (see image below) which seems to indicate there is 1 CAN controller with Jetson Orin Nano, but because I am not a hardware person I cannot tell if this limitation extends to devices attached over USB. If that is not possible, could you suggest any alternate solutions?

[image]

Thanks in advance for your help!

RosenYin, 2025-03-03:
这个问题我们也遇见了，暂时没有软件解决的办法，只能在内核编译阶段将模块编译进去。

这个问题在nvidia论坛有人遇见了，是因为内核没有gs_usb模块，你需要使用命令行的Jetpack ，然后修改6.0(或以上)的txt，参照如下<https://forums.developer.nvidia.com/t/missing-gs-usb-kernel-module-for-jetpack-6/275287/7>，增加`CONFIG_CAN_GS_USB=m`等，也就是参照`defconfig.jetpack_5_1_2.txt`文件，增加`defconfig.jetpack_6_0.txt`中缺少的，补全有关can的gs_usb模块，然后重新编译内核。

notmahi, 2025-03-10:
Thank you, setting up `GS_USB` by re-flashing the Kernel was the solution. I appreciate the help.

closed 2025-03-10 by notmahi