agilexrobotics/piper_sdk #32: piper sdk version

Crypto69, 2025-03-23:
The read me file says:

pip3 show piper_sdk
0.0.x provides SDK support for robotic arm firmware versions prior to V1.5-2.

My hardware says s-v 1.5-8

when i run 
piper@agilex:~/agilex/Piper_sdk_ui-master$ pip3 show piper_sdk
Name: piper-sdk
Version: 0.0.9
Summary: A sdk to control piper
Home-page: UNKNOWN
Author: RosenYin
Author-email: yinruocheng321@gmail.com
License: UNKNOWN
Location: /home/piper/.local/lib/python3.10/site-packages
Requires: python-can
Required-by: 

My SDK version is 0.0.9

So what does this mean because the read me says 0.0.x provides SDK support for robotic arm firmware versions prior to V1.5-2. but firmware is
s-v 1.5-8

If I try to run any of the V2 demos or the ui i get
piper@agilex:~/agilex/Piper_sdk_ui-master$ /bin/python3 /home/piper/agilex/Piper_sdk_ui-master/piper_ui.py
Traceback (most recent call last):
  File "/home/piper/agilex/Piper_sdk_ui-master/piper_ui.py", line 11, in <module>
    from piper_sdk import C_PiperInterface_V2  
ImportError: cannot import name 'C_PiperInterface_V2' from 'piper_sdk' (/home/piper/.local/lib/python3.10/site-packages/piper_sdk/__init__.py)

What does this mean? How do I solve this problem?

Crypto69, 2025-03-24:
I managed to fix this by doing an uninstall
And selecting the version of the package to install .. piper@agilex:~$ pip3 show piper_sdk
Name: piper_sdk
Version: 0.2.19
Summary: A sdk to control Agilex piper arm
Home-page: https://github.com/agilexrobotics/piper_sdk
Author: RosenYin
Author-email: yinruocheng321@gmail.com
License: MIT License
Location: /home/piper/.local/lib/python3.10/site-packages
Requires: python-can

I still have another question. What is the difference between the demos in version one in version 2 folder? 

Is this something to do with the firmware? My hardware version is s-v 1.5-8 does that mean I must use version 1 demo ?

RosenYin, 2025-03-27:
在CHANGELOG中的0.1.12版本有对V1和V2的说明，并且在此后的版本也遵循这个修改规则：
[image]

closed 2025-03-31 by HoranTian