agilexrobotics/piper_sdk #35: Inconsistent Joint States on Piper Arm at Startup

aman-pandey23, 2025-04-04:
We’ve observed inconsistent behavior with one of our Piper robotic arms, where joint states are incorrectly initialized on startup and require a series of steps to correct. The issue is primarily visible on joint2, which reports an unusually high value near its zero position. Following this, the arm attempts to move to an incorrect zero pose when started, and even after manually setting the zeroes via the Piper SDK UI, joint state mappings remain incorrect. This results in some erratic behavior, where moving joint1 affects joints 4–6 in some random manner and same with other joint states. A workaround is required to restore correct joint values, which involves toggling between Master mode and Slave mode, followed by a reset.

- On startup, joint2 reports an abnormally high value near its zero position.
- The arm attempts to move to an incorrect, random zero position.
- Setting zeroes via the Piper SDK UI doesn’t directly resolve the issue.
- Joint state readings remain incorrect after setting zeroes.
- Moving joint1 causes random changes in joints 4–6 and similar random mapping from one joint to multiple.
- Workaround is to Switch to Master mode, then back to Slave mode and perform a reset whcih then shows correct joint values

utkarsh-maniro, 2025-05-05:
Hi, I am facing a similar issue in the piper arm(https://github.com/agilexrobotics/piper_sdk/issues/37), I followed the master to slave mode switch, but it does not work for me. Did you try anything else that worked?

RosenYin, 2025-05-12:
这很大概率是因为夹爪助力等级被设定为255导致，你可以查看上位机的夹爪设定部分，然后将助力等级设定为1

CodeHesed, 2026-02-06:
@RosenYin I tried to find the "夹爪设定部分", but I couldn't find it. May I ask which part have you actually meant?

CodeHesed, 2026-03-09:
Having a same issue here. Thinking about whether I should do a recalibration every reboot.

CodeHesed, 2026-03-22:
> I am facing the same problem, 
> when i record data of piper using master-slave mode. 
> and then replay the piper without turning off the power, the performance is good with high repeat rate (10/10). 
> However, when i turn off the power and replay the same data. It does not work anymore. 
> The position is shifted. 
> Any solution to this?

I just found out that initial state of joint 1-3 at initialization affects the robot arm's movement.

It worked better for me whenever I visually check joint 1 fitting to the guidance lines. 
(Joint 2 and 3 can be initialized on it's natural state, hanging down due to gravity)

But still, I need higher precision 🥲