agilexrobotics/piper_sdk #73: [Question] Proper torque scaling factors for MIT control

kwongeunwoo, 2025-08-19:
Hello,

I am currently trying to control the robot using piper_sdk in MIT mode.

While attempting gravity compensation with piper_ros using the piper_no_gripper_description.urdf, I encountered an issue.
When I directly apply the gravity torque values computed from the URDF, the joints produce excessive torque and result in abnormal movements.

Through testing, I found that multiplying the torques by:

0.118125 (= _COEFFICIENT_1_3 / 10) for joints 1, 2, and 3

0.47922 (= _COEFFICIENT_4_6 / 2) for joints 4, 5, and 6

resulted in proper gravity compensation.

Since these values were obtained empirically, I would like to ask:
What are the correct scaling factors for each joint’s torque in order to achieve accurate torque control?

Thank you.

RosenYin, 2025-10-15:
了解到您的机械臂固件版本是1.5-8，在这个版本下

发送目标力矩，电机实际执行的力矩为输入数值的4倍，例如发送1N，电机执行4N；

对于反馈力矩，j1-3关节反馈力矩=电流\*1.18125/4，j4-6关节反馈力矩=电流\*0.95844；

在sdk中使用 `GetArmHighSpdInfoMsgs().motor_1.effort` 可以得到电机1的反馈力矩，修改`motor_x`中的`x`可以获取1-6关节电机的反馈力矩，这个数值是由上述公式计算得到，只不过其中j1-3关节反馈力矩没有除以4，因此对于j1-3关节，需要自行将这个数值除以4才是真实反馈力矩，对于j4-6,sdk计算的反馈力矩不需要再进行其它处理。

----
I understand your arm's firmware version is 1.5-8. In this version,

When you send a target torque, the actual torque the motor executes is 4 times the input value. For example, if you send 1N, the motor executes 4N.

For feedback torque, the feedback torque for joints J1-3 = current * 1.18125 / 4, and the feedback torque for joints J4-6 = current * 0.95844.

In the SDK, use `GetArmHighSpdInfoMsgs().motor_1.effort` to obtain the feedback torque of motor 1. Modify `x` in `motor_x` to obtain the force feedback torque for joints 1-6. This value is calculated using the above formula, except that the feedback torque for joints J1-3 is not divided by 4. Therefore, for joints J1-3, you need to divide this value by 4 to obtain the actual feedback torque. For joints J4-6, the SDK-calculated feedback torque does not require any further processing.

kwongeunwoo, 2025-10-15:
Thank you for the answer. I will apply it and test it out.